yaapu/HorusMappingWidget #2: No telemetry Data

dentex, 2020-03-31:
Ciao Alessadro, 
I was trying to setup your widget; my setup:
_Jumper T16 radio
R9M module
R9MM receiver
**FPort**;
Betaflight_

The telemetry works, I have all the data on the radio:
[image]

The widgets are not showing anything. 
[image]
[image]

As a "secondary" problem I also do not have maps, but I'm sure I loaded the correct tiles.
[image] 
I checked for my location and its tiles manually and they are there on the sdcard.

I'm sure it's something on my side.
Do you have any suggestion?
Please let me know if you need more info.

Grazie.

yaapu, 2020-03-31:
Hi the Full Telemetry Widget does not work with Betaflight, it requires ardupilot.
The Mapping Widget does not find the tiles so you probably did something wrong with the tiles.
The tiles engine is used by several on the full widget so the most probable cause is an error in your tiles db, please follow the instructions step by step:
- download (the correct zoom levels, from your screenshot I'd say -2,-1 and0)
- resize
- copy

dentex, 2020-03-31:
OK, so I have to use the Mapping widget only. Great.
For the map tiles... I'll check one more time. 

Thanks.   :)

dentex, 2020-03-31:
I found my way through it!
a)The first time I copied the file on the sdcard, probably something got corrupted, or I don't know...; so I though I needed also the FrSkyTelemetry script (that instead supports Ardupilot only)... 
b)And for the tiles, that's another story. I downloaded/edited them in Linux, under wine... it turned out the script didn't like it. 

So:
fix-a) I just deleted everything and copied again the script files (Mapping widget only) and;
fix-b) I did again the tiles work in VirtualBox (Win7 machine).

That finally worked:

[image]

Thanks Alessadro for the wonderful job.

yaapu, 2020-03-31:
Great news!

closed 2020-03-31 by yaapu