yaapu/HorusMappingWidget #3: iNAV: add support for displaying home and calculate home distance and bearing

yaapu, 2020-04-02:


lossUAVes, 2020-07-15:
Hi, I love this widget. Thank you.
I'm trying to run it in iNav, but I only have No telemetry data message.

INAV 2.5.1. FPort serial protocol.
Holybro Kakute F7 mini flight controller.
Frsky R-XSR receiver. Firm: RXSR_ACCST_2.1.0_Fport_LBT
RadioMaster TX16S transmitter. Firm: multi-stm-opentx-aetr-noinv-v1.3.1.36
OpenTX 2.3.9. Firm: opentx-tx16s-lua-noheli-es-2.3.9-otx
SD Card contents: sdcard-480x272-2.3V0029
Yaapu telemetry widget 1.8.1.

INAV telemetry lua script is working fine.

yaapu, 2020-07-15:
Whick widget are you trying to run? The Yaapu Telemetry Widget or the HorusMappingWidget?
The former does not work with iNAV and only support ArduPilot!

lossUAVes, 2020-07-16:
Ah, sorry... I didn't know that there was two widgets... HorusMappingWidget is working. Thanks.

yaapu, 2020-07-16:
no worries :-)

yaapu, 2022-11-27:
added in version 2.0.0

closed 2022-11-27 by yaapu