yaapu/HorusMappingWidget #6: [suggestion] dummy GPS fix for testing

dentex, 2020-04-05:
Hello,
I don't know if this is somehow already implemented, but for the moment, if you want to see how the tiles and the sensors panes look like, you have to have a valid GPS fix; a real one, with the drone outside, plugged LiPO and all.
I was thinking about something like a simple text file, with fictional GPS coordinates in it; when the script finds it, it may show that fix instead of waiting for the real one. Or maybe just add an option where the user can type in the coordinates for a testing purpose GPS fix.

Thanks.

yaapu, 2020-04-05:
Hi, you can do that in companion, install the widget on the simulator, enable sensors simulation and enter the gps coordinates there.

dentex, 2020-04-05:
Great! Thanks. 
I didn't know.

dentex, 2020-04-06:
Hey Yaapu, 
I was trying to run the script from the radio simulator. 
It seems all is set up (I have sounds, etc), but the telemetry page with the map widget won't show up.
Looking for your script in the SYS menu, I have this error:
[image]

yaapu, 2020-04-06:
Hi, download latest version again, you hav the wrong mapsconfig.lua.

Never seen that error before, is it on linux?
Try to use [these files](https://github.com/yaapu/HorusMappingWidget/tree/master/HORUS/SOURCES/SRC) and let companion compile it for you

dentex, 2020-04-06:
Yes Linux.

It seems you have to add again the telemetry script at every simulator startup... but then it works.

[image]

No sensors here, but I'll try to figure it out. 
:)

this is my sensors setup file in `SCRIPTS/YAAPU/CFG/r9mmaps_sensors.lua`

```----------------------------------------
-- custom sensors configuration file
----------------------------------------

local sensors = {
  -- Sensor 1
[1]=  {
    "RSSI",   -- label
    "RSSI",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "dBm",      -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    40,        -- warning level (nil is do not use feature)
    25,        -- critical level (nil is do not use feature)
  },
  -- Sensor 2
[2]=  {
    "BattV",   -- label
    "VFAS",     -- OpenTX sensor name
    2,          -- precision: number of decimals 0,1,2
    "C",        -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },

  -- Sensor 3
[3]=  {
    "CellV",   -- label
    "A4",      -- OpenTX sensor name
    2,          -- precision: number of decimals 0,1,2
    "C",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },

  -- Sensor 4
[4]=  {
    "Curr",   -- label
    "Curr",     -- OpenTX sensor name
    1,          -- precision: number of decimals 0,1,2
    "A",   -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },

  -- Sensor 5
[5]=  {
    "Temp1",   -- label
    "Tmp1",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "C",        -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },

  -- Sensor 6
[6]=  {
    "GSpd",   -- label
    "GSpd",     -- OpenTX sensor name
    1,          -- precision: number of decimals 0,1,2
    "kt",        -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
}
------------------------------------------------------
-- the script can optionally look up values here
-- for each sensor and display the corresponding text instead
-- as an example to associate a lookup table to sensor 3 declare it like
--
-- [3] = {
--   [-10] = "ERR",
--   [0] = "OK",
--   [10] = "CRIT",
-- }
-- this would display the sensor value except when the value corresponds to one
-- of entered above
-- 
local lookups = {
  -- lookup 2
  --[[
  [6] = {
      [-30] = "ERROR",
      [-20] = "OFF",
      [-10] = "COOL",
      [-1] = "LOCK",
      [0] = "STOP",
      [10] = "RUN",
      [20] = "REL",
      [25] = "GLOW",
      [30] = "SPIN",
      [40] = "FIRE",
      [45] = "IGNT",
      [50] = "HEAT",
      [60] = "ACCE",
      [65] = "CAL",
      [70] = "IDLE",
  },
  --]]
}

collectgarbage()

return {
  sensors=sensors,lookups=lookups
}```

dentex, 2020-04-06:
[image]

Got it!

Thanks again.

closed 2020-04-06 by dentex