yaapu/HorusMappingWidget #8: Arrow is often close to the boarder of the map

StefanKellerAC, 2020-04-13:
You switch the tiles when the arrow leaves the map, I like it more to switch if it is outside a border.
the following change does the job in mapslayout.lua line ~420:

`      local Border = 50`
`      local myCode = drawLib.computeOutCode(myScreenX, myScreenY, minX+17+Border, minY+17+Border, minX+17+ 4*100 - Border, minY+17+ 2*100 - Border);
`

a setting would be fine of course :-)

yaapu, 2020-04-14:
I switch tiles when the arrow touches the border, if your setup is behaving differently please provide a video, I cannot reproduce what you describe, and IMO a setting for this is just overkill.

StefanKellerAC, 2020-04-14:
You switches tiles, when the arrows **leaves** the map, so it can be a long time close to the border of the map,
With my change it switches when it comes closer than 50 to the border of the map.
So it is centered better.

I didnt change your function loadAndCenterTiles which selects the tiles, it is just called earlier respectively more often.

Maybe a value of 100 for x would be even better, than the arrow would never be in the left- and rightmost tiles...

yaapu, 2020-04-14:
Stefan that is not true, please take a look at this [video](https://youtu.be/tyHzxsb5Tio), the tile switch is at 1/2 of the triangle vehicle diameter, so as soon as it touches the border I reload the tile

StefanKellerAC, 2020-04-14:
hmm, in your video the behavior is fine, 
but I cant understand why!
It was different for me!

your function loadAndCenterTiles is only called if
myCode > 0
which is calculated by:
`      local myCode = drawLib.computeOutCode(myScreenX, myScreenY, minX+17, minY+17, maxX-17, maxY-17);
`
btw: do I have to use the source code in the zip from Folder PP or from SRC ??
what is the difference?

yaapu, 2020-04-14:
`local myCode = drawLib.computeOutCode(myScreenX, myScreenY, minX+17, minY+17, maxX-17, maxY-17);`

17 is the triangle vehicle radius,-myCode is > 0 when the vehicle center is inside a border of 17 pixels (the border is inside the tiles of cource)

Use the SRC for your testing, the PP needs a lua preprocessor to process the #defines I use for conditional compiling and constants

StefanKellerAC, 2020-04-14:
ok I mixed up this 17 with your
`local minY = 18`
which is used for the header

but than it is clear: with your border of 17 the arrow can nearly touch the "end of the world" :-) for a longer time, when flying parallel to it, which is what I have seen. 

Now I would like more or less this version:
`      local myCode = drawLib.computeOutCode(myScreenX, myScreenY, minX+50, minY+80, maxX-80, maxY-50);`

disadvantage with my setting is, that the maps jumps more often

yaapu, 2020-04-14:
This pretty much fixes all pending issues, 
- default sensor file
- timer 3 reset
- switch tile sooner
- widget reset on model change

please test it and report back

[HorusMappingWidget-1.1dev.zip](https://github.com/yaapu/HorusMappingWidget/files/4476487/HorusMappingWidget-1.1dev.zip)

StefanKellerAC, 2020-04-14:
yes looks fine, thx!

next step would be some kind of hysteresis, because when you circle in thermals at the switching border the maps switches frequently :-)
but I think in the field this is not so important because then you will not look on the display :-)

StefanKellerAC, 2020-04-18:
hysteresis is easy to implement:
just set the border to for example 40 instead of 50.
when the tile changes the arrow will then be 60 away from the border of the new map!

closed 2020-04-25 by yaapu