yaapu/HorusMappingWidget #9: no default sensor settings

StefanKellerAC, 2020-04-13:
please load a default sensor definition file, when no specific one is found.
eg: default_sensors_maps.lua

yaapu, 2020-04-14:
Hi, there's no such thing as a default sensors file, each controller exposes telemetry with different sensor names. You're welcome to contribute examples for your specific use cases.

StefanKellerAC, 2020-04-14:
I fly gliders and guess I will use for all or most of them the same layout.
to avoid multi copies of this file I made a default file with that layout.

Therefore I changed loading this way:

```
utils.loadCustomSensors = function()
  local success, sensorScript = pcall(loadScript,getSensorsConfigFilename())
  if sensorScript == nil then
    success, sensorScript = pcall(loadScript,"/SCRIPTS/YAAPU/CFG/default_sensors_maps.lua")
  end
```
another question:
what is the difference between ckecking success and sensorscript == nil  ?

yaapu, 2020-04-14:
ok, this makes much more sense, if a model specific is not found use default, ok I like this approach.

StefanKellerAC, 2020-04-14:
lua is better to understand than text :-)

closed 2020-04-25 by yaapu