yaapu/HorusMappingWidget #12: Need battery remaining capacity through voltage.

Moksh-2000, 2020-04-22:
I know there is such a big jok , but listen I was tryed very time and I successfully processed (for me = in my case) 
https://discuss.ardupilot.org/t/battery-estimate-percentage/55254
This is my invoice , also this is a video show when battery not connected to dron also not in load then battery voltage and when battery connect to drone then battery voltage not change (not a big difference) you can see in this video. So my resume is when connect battery then calculate battery voltage also calculate battery percentage (mention in my ardupilot server , how to do that) and continue with current "Amps" compensation. It mean we have "Estimate" battery percentage and start with normal calculation. I hope you understand what I'm talking about.

https://photos.app.goo.gl/TdRp7zwcoyXQiueC9

Need :- can you make a new sensor (in script) 【for me right now , for beta [atleast my satisfaction] script】

Thank you 🙂

yaapu, 2020-04-22:
Hi, sorry but this is a mapping widget not a telemetry widget, I draw sensors already present and won't implement new sensors :-(

Moksh-2000, 2020-04-23:
Ok sir no problem ☺️ your work also impressive 😍.  And sir Who should I meet for this job?
And sir last 
[image] this page i love it but there is no more pages like previous one and I added some things about them but I can't figure out more 4 pages so can you replace with this page to this page ?
[image]

Thank you 🙂

yaapu, 2020-04-23:
Hello,
Why are you running the mapping widget in a page and the telemetry widget in another page?
The telemetry widget already has the mapping feature!

If you're running the yaapu telemetry widget my latest dev version (not released yet) has battery % by voltage as a new feature

Moksh-2000, 2020-04-23:
Ohh yeah that really interesting. Yes my need is battery percentage showing
by battery voltage is batter , and I'm talking to you I need this page
exchange with page 2 because I need more information about map data ,
screen 1 also good but I need bigger map size 😁

Thank you 🙂

closed 2021-04-18 by yaapu