yaapu/HorusMappingWidget #14: Widjet crash\disabled on Inav and FrSky D8 telemetry

7ALHIMIK7, 2020-04-23:
OpenTx sensor "Cels" from Inav and FrSky D8 disable widjet after first telemetry packets. Same config but without this sensor works fine.
Is it possibly to disable warning message "No telemetry" in front of the map? In case you have lost UAV this message blocks map and it is immosible to check last position.
Thank you for the great widget!

yaapu, 2020-04-25:
You can't use the cels sensor because it's a combined table sensor, you need to manually define each individual cell as a calculated sensor and use those in the widget, take alook here for an example https://www.youtube.com/watch?v=l8TeVq1bj4Y

yaapu, 2020-04-25:
I fixe the bid "no telemetry" dialog issue

closed 2020-04-25 by yaapu