yaapu/HorusMappingWidget #20: Horusmappingwidget t16

beachbum40, 2020-10-13:
Can't seem to get the 10 telemetry blocks on screen right side and bottom.  I have all sensors detected and working just can't show on screen. read wiki but confused to say the least on how to configure.  Any help would be appreciated thanks.

Papsi77, 2020-10-15:
https://github.com/yaapu/HorusMappingWidget/wiki/user-selected-sensors-setup

realy easy...

beachbum40, 2020-10-16:
For you maybe, Not for me.   Lol.  I've since decided against the maps on
my T16 pro. Map resolution is horrible and I can barely see where I'm at.
I fly strictly country side with no roads to so this make it even more
difficult.   I've already got a bluetooth mod from R9M and Have full XF
which has inbuilt bluetooth and I can send telemetry to Apps like Telemetry
Viewer which has all the mapping I need plus telemetry sent over
bluetooth.  The dev is also working on having offline map going as well
which h will help when I'm flying in areas without cell service which is
often but think this is best route for me.  Plus I use Inav lua scripts
which gives nice sharp HUD with home point direction and all telemetry.
 Yaapu is nice but way to much involved in setting up with inav imo and
resolution on Most if not all these TX'S just isnt there to make this
useful in my opinion.

closed 2020-10-16 by yaapu