yaapu/HorusMappingWidget #21: Widget becomes disabled on arm, on Radiomaster TX16S / Crossfire

Batperson, 2021-01-10:
Hi,

I am attempting to run the mapping widget on a Radiomaster TX16S with Crossfire, with a Tyro129 quad. Since upgrading from OpenTX 2.3.10 to 2.3.11 the widget becomes disabled when the quad is powered on

Steps to reproduce:
1. Turn on radio
2. Switch to map widget
3. Connect battery on quad
4. Observe some telemetry values appear
5. ESC beeps
6. Widget is disabled

OpenTX 2.3.11 fixed an issue related to Crossfire where the FM flight mode sensor was never updated, not sure if this is related or not.

yaapu, 2021-01-21:
Hi, I'd say one of your configured sensors is crashing, delete you sensors file, and run the map only.
Create a sensor file with just 1 sensor and activate them 1 at the time until you find the one crashing, than we can further investigate why.

Batperson, 2021-04-10:
I suspect it was the FM sensor, after removing this from the config it is working again.

Eg681, 2021-04-16:
So your running this script with radiomaster tx16s and crossfire, just curious what version of inav are you using?

Batperson, 2021-04-17:
Yes that's correct. I'm on iNav 2.6.

closed 2021-04-18 by yaapu