yaapu/HorusMappingWidget #26: Yaapu telemetry

Eg681, 2021-02-23:
I've tried everything to get telemetry working on yaapu maps but still no go, I've created the model_sensor_maps.lua in yappu cfg page copied in sensors page but will not show up on maps, I'm using radiomaster tx16s, crossfire micro tx v2 and nano receiver, f405-wing flashed with inav. All sensors working on radio I've deleted and rediscovered a couple times but still no go. Opentx 2.3.9 and inav 2.5.0 what else can I try to get this working? Thanks

yaapu, 2021-02-24:
please attach both your maps cfg and sensor files

Eg681, 2021-02-25:
----------------------------------------
-- custom sensors configuration file
----------------------------------------
local sensors = {
  -- Sensor 1
  -- Sensor 2
[1]=  {
    "Alt",   -- label
    "Alt",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "m",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    50,        -- warning level (nil is do not use feature)
    100,        -- critical level (nil is do not use feature)
  },
  -- Sensor 6
[2]=  {
    "Spd",   -- label
    "GSpd",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "m/s",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
  -- Sensor 6
[3]=  {
    "Dist",   -- label
    "Dist",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "m",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
[4]=  {
    "VSpd",   -- label
    "VSpd",     -- OpenTX sensor name
    1,          -- precision: number of decimals 0,1,2
    "m/s",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
[5]=  {
    "Hdg",   -- label
    "Hdg",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "@",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
[6]=  {
    "Celd",   -- label
    "Celd",     -- OpenTX sensor name
    2,          -- precision: number of decimals 0,1,2
    "V",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    3.65,        -- warning level (nil is do not use feature)
    3.30,        -- critical level (nil is do not use feature)
  },
[7]=  {
    "Celm",   -- label
    "Celm",     -- OpenTX sensor name
    2,          -- precision: number of decimals 0,1,2
    "V",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    3.65,        -- warning level (nil is do not use feature)
    3.30,        -- critical level (nil is do not use feature)
  },
[8]=  {
    "Batt",   -- label
    "VFAS",     -- OpenTX sensor name
    1,          -- precision: number of decimals 0,1,2
    "V",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },
[9]=  {
    "Fuel",   -- label
    "Fuel",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "%",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    50,        -- warning level (nil is do not use feature)
    25,        -- critical level (nil is do not use feature)
  },
[10]=  {
    "Curr",   -- label
    "Curr",     -- OpenTX sensor name
    1,          -- precision: number of decimals 0,1,2
    "A",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "-",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)

Eg681, 2021-02-25:
MAPZ:1,MAPT:1,MAPG:2,MAPTD:10,ZTC:32           This is whats in maps file.

Eg681, 2021-02-25:
my sensor page in the cfg that i created i've named sonic1213_sensors_maps.lua
I have only one model setup in this radio, maps work great 3 zoom levels no problems there Im showing all sensors in telemetry page of radio and i have deleted all sensors and then rediscovered but still no go. I hope you can help. Thanks

Eg681, 2021-02-25:
I've sent you what you asked for I hope this will help resolve the issue,
and I hope I didn't close the issue by accident I'm new to this so if did
I'm sorry.

yaapu, 2021-02-25:
please attach the files, not the content, I need to check file names as well.

anyway the sensor configuration file is missing closing braces at the end:

`}}`

Eg681, 2021-02-25:
[CFG.zip](https://github.com/yaapu/HorusMappingWidget/files/6043246/CFG.zip)

Eg681, 2021-02-25:
here is all CFG Files

Eg681, 2021-02-25:
[CFG.zip](https://github.com/yaapu/HorusMappingWidget/files/6043595/CFG.zip)
I made changes to sensor page, added brackets but still no telemetry on map borders.

Eg681, 2021-02-26:
I tried this sensor file and renamed it but still not working, please look
it over and tell me whats wrong, I know i doubled the sensor page just
grasping at straws but that dont work

Eg681, 2021-02-26:
I erased radio sd card and reinstalled everything from my backup files,
installed maps and renamed and installed sensor page, I also
rediscovered sensors on radio everything looks good there. still no sensor
information on maps, everything the same no changes at all. Please look
over CFG files  I must be missing something. Thanks

yaapu, 2021-02-26:
Hi, file name is wrong, correct file names should be:
- sonic1213.cfg
- sonic1213_maps.cfg
- sonic1213_sensors_maps.lua

Eg681, 2021-02-27:
Ok thanks, I'll make the changes and go from there

Eg681, 2021-02-27:
I made changes to CFG files like you recommended but no change at all, I do
notice that the 2 1k files look  to be the exact same thing but maybe thats
normal. I've attached the new CFG file please take a look I would love to
get this program working. Thank you

yaapu, 2021-02-27:
sonic1213_sensors_maps.lua this file should NOT end with .cfg but with .lua, plese confirm this

Eg681, 2021-02-27:
https://linksharing.samsungcloud.com/wR52yPD0Dk2e

On Sat, Feb 27, 2021, 1:27 PM Eg Treadway <egtreadway@gmail.com> wrote:

> I can confirm it ends in .lua
>

yaapu, 2021-02-28:
nope same issue, type of sonic1213_sensors_maps.lua is reported as CFG, should be lua, this means that the real file name is sonic1213_sensors_maps.lua.cfg and not sonic1213_sensors_maps.lua

Eg681, 2021-02-28:
https://linksharing.samsungcloud.com/wR52yPD0Dk2e- If I type in sonic1213_sensors_maps.lua why is it reporting as cfg, sorry I don't understand do I need to change model name in radio and then change in cfg files, please explain

HazzaHFPV, 2021-02-28:
Have you enabled displaying files extensions in Windows? There might be .cfg at the end as extension but not displayed.

https://www.google.com/amp/s/www.howtogeek.com/205086/beginner-how-to-make-windows-show-file-extensions/amp/

Eg681, 2021-02-28:
Thanks for this link, I'm using windows 7 pro and I think this its what's happening because it's not showing me file type.

Eg681, 2021-03-01:
i've enabled show file type so at least I can now see that thanks to HazzaHFPV, I've double checked my sd card yaappu cfg files but still no sensors information on maps.
[cfg zip.zip](https://github.com/yaapu/HorusMappingWidget/files/6061836/cfg.zip.zip)

Eg681, 2021-03-01:
I'm closing this issue now that everthing is working, on my sensor page which i named sonic1213_sensors_maps.lua i've been days trying to get sensor information on the map borders but finally I checked properties of this file and it clearly showed at the end that .lua was doubled, it showed .lua.lua on file name i created in yaapu cfg page it showed only one .lua at the end, I deleted .lua from this file and Boom maps and sensors working. Maybe this will help someone else, thanks

yaapu, 2021-03-02:
glad you finally made it work!

Eg681, 2021-03-04:
I've been playing with this program some and it's amazing I have to say, only thing is I can't get any sound working with the telemetry, I've tried going to yaapu sounds on radio sd card and choose english.psv but there's no excute opition, What am I doing wrong? Thanks

yaapu, 2021-03-05:
Hi, there are no sounds or alerts, this widget is mainly for mapping, sensor display was added as a "bonus" don't expect alerts such as those found in my ardupilot specific widget!

Eg681, 2021-03-05:
Thank you for the clarification.

Eg681, 2021-03-11:
I'm trying to figure out why the plane symbol on the map is not working
proper while flying, sometimes it's not moving sometimes it's going side
ways not pointing in correct direction, the hdg sensor changes as I spin
the plane changes in degrees,  any help will be appreciated.

Eg681, 2021-03-20:
I enjoy flying with this lua script but still trying to figure out why the hdg. Sensor is not working proper, there's settings in radio that change the way the sensor reports back to radio but correct settings are beyond me. Sensor will change in degrees as I spin the plane but while flying it's not going in correct direction and flying sideways. What am I missing here? Any help will be appreciated.

Eg681, 2021-04-13:
I'm still trying to figure out why the pointer on map won't show correct,
with the help from others it seems that this script is not going to work
with crossfire because of the way it determines heading- bearing, is this
correct? Others say that if running inav lua and yaapu on same sd card that
the two will clash and not work correctly is this true? I've been in touch
with 2 Youtubers that say yaapu is working with inav and crossfire no
problem. If the developer would chime in it would be alot of help. Thanks

yaapu, 2021-04-13:
My widget works fine with crossfire and inav, this is an issue with your setup.
The widget expects a sensor named "Hdg" (case is important) with a value in degrees, you can set it up in companion simulator to verify it works.

[image]

Eg681, 2021-04-13:
Ok thanks maybe maybe my h in hdg not capitalized I'll check it out

Eg681, 2021-04-14:
Just to let you know, there's alot of misconception about your script,
Youtubers and others have tried to help me the best they could but
apparently just don't know  like me, but again thanks

Eg681, 2021-04-14:
https://youtu.be/yZpgoB1SRGY This is short video showing my sensors name
and that first letters are capitalized, gps coordinates are wrong and Hdg
sensor wrong also arrow on map not spinning when I spin plane, in video I
switch back and forth between inav and yaapu to show difference, I hate
bugging you with this I'm sure your a busy man but I'm out of opitions? I
will be happy to donate to the cause I just want to get this working.
Thanks

yaapu, 2021-04-15:
Hi, from the video the widget is rotating the vehicle using the Hdg value allright since the rotation matches the Hdg value of both the discovered Hdg sensor and the Hdg sensor displayed right of the map, what I do not understand is why the iNAV lua is rotating using a value of 133, it's like it's not using Hdg but rather another value. I'm not an iNav user so I simply don't know the answer.
And you're not bugging me for sure, it's just hard to believe that there's an issue in the heading code, I would have had way more reports of such an isuue if that was the case.
Donations are for happy users, not to have me solve problems :-)

Eg681, 2021-04-15:
Thanks for response, I should have explained better but in video looking at
laptop screen craft symbol is pointing at landmark that I know is 137
degrees from my location I have flew there many times and double checked
with hand compass, you can see the same landmark on the yaapu map but arrow
pointing wrong direction, also inav gps coordinates is matching Google maps
coordinates but yaapu coordinates numbers big difference, if I spin plane
hdg numbers on yaapu maps change some but never more than 30 to 35 degrees,
I also know 0 degrees and 180 degrees direction according to land Mark's
were I fly but arrow on maps not pointing in correct direction. Just trying
different things I changed yaw sensor to degrees in radio and pointer on
map spins randomly on map but still not in correct direction, this may not
be useful information. I would love for this to work the way it's intended
but still no go. Thanks again

yaapu, 2021-04-18:
closing as has become a duplicate of https://github.com/yaapu/HorusMappingWidget/issues/27

closed 2021-04-18 by yaapu