yaapu/HorusMappingWidget #27: Hedge sensor

Eg681, 2021-03-07:
I've been flying with this program and it's great, the only thing I can't get worked out is the plane symbol on map while I'm flying it's going in correct direction but its flying side ways not pointing in direction its actually going, in my inav lua the plane symbol is always pointing in correct position so I don't think it's a sensor problem but not sure. Any help will be appreciated

Eg681, 2021-04-02:
I'm still trying to get the hdg sensor working proper, is the bearing of the plane symbol on maps getting information from the hdg or yaw sensor? If I change any settings in radio for these sensors the yaapu telemetry works somewhat better but then the inav lua telemetry such as bearing is all over the place. I'm assuming the problem is in the yaapu cfg sensor page that I created but all other sensors seem fine. Any advise would be great

Eg681, 2021-04-06:
https://youtu.be/5AoCaZn4lJo) this is link to video showing the difference in sensors like hdg and fuel between inav lua and when on yaapu maps telemetry page, also arrow on maps not pointing in correct direction. Why is sensors showing different information between inav telemetry page and yaapu? Any help will be appreciated, Thanks

HazzaHFPV, 2021-04-06:
In my eyes yaapu shows the correct value and arrow shows roughly 30°?
 Seems to be an issue with the inav telemetry script. I recommend raising the issue there. Have you followed the setup guide for the inav telemetry lua? Set correct unit and capacity for fuel indication?

Eg681, 2021-04-06:
I've been flying with inav for last 2 years and the hdg sensor should show 0 degrees at due north and 180 degrees at due south which inav lua is always correct, yaapu script doesn't it always shows numbers in degrees but numbers not even close, also looking at video fuel in inav is 95% which is correct yaapu showing 56%. This could be a crossfire issue I'm not sure. If anyone else has dealt with this please post. Thanks

Eg681, 2021-04-06:
How could it be a unit or capacity setting if everything is correct in inav lua?

MrD-RC, 2021-04-06:
The INAV LUA is correct. The issue is the sensors being used. 

With SmartPort (which works as expected) the Hdg sensor is actually the yaw of the aircraft in decidegrees. (INAV telemetry/smartport.c line 435).

However, with Crossfire telemetry; the heading sensor is the ground course of the aircraft. The heading on the aircraft is the Yaw sensor in Crossfire telemetry, in radians (INAV telemetry/crsf.c line 300).

So the sensor being used to determine the direction the aircraft is facing is incorrect with Crossfire telemetry sensors. 

The fuel could be setup. @Eg681 I’m thinking the INAV LUA hasn’t been configured correctly for the fuel type. The maps LUA matches the sensors screen.

Eg681, 2021-04-06:
Ok that makes sense, so how do I setup the yaw sensor for when I'm viewing the yaapu maps and and is it possible to scroll between inav lua and yaapu during flight? Or do I have to use one or the other? Are you using yaapu, inav and crossfire?

Eg681, 2021-04-06:
It just seems that if I change yaw or any sensor in radio that it would change inav lua, maybe it's not possible to use yaapu, inav lua and crossfire I've done alot of YouTubeing and reading but have found no confirmation. Any help with how to get both or at least the yaapu maps working would be great

MrD-RC, 2021-04-07:
@Eg681 this would need to be fixed in the LUA script. There’s nothing you can do in OpenTX.

Eg681, 2021-04-07:
It's making sense now, the yaapu script should be using yaw sensor in radians but it's not therefore the arrow on maps won't be correct so that would need fixed in yaapu script is this correct? One last question while flying will the arrow be in correct location on map but pointing wrong direction? If so I could use inav lua on return to verify direction. Maybe yaapu will chime in on this issue but thanks for all your research and help

yaapu, 2021-04-18:
ok this finally makes sense, thanks @MrD-RC .
I could add a menu option to select between "Hdg" and "Yaw" as sensor to use, how's that?

yaapu, 2021-04-18:
@Eg681 please try this version, enter the config menu and select "Yaw" as heading sensor, also select "rad" as heading unit,hopefully it should work as expected
[HorusMappingWidget_1.2.1dev.zip](https://github.com/yaapu/HorusMappingWidget/files/6331107/HorusMappingWidget_1.2.1dev.zip)

Eg681, 2021-04-18:
Ok thanks I will give it a try

yaapu, 2021-04-25:
hi, did it fix your issue?

Eg681, 2021-04-25:
Thanks for checking but haven't tried yet, to be honest I'm waiting for MRD-RC to test it with crossfire and post his results but I do have high hopes. I will definitely post here with final results.

Eg681, 2021-04-28:
I've copied over the new script 1.2.2 b and did change heading sensor to yaw and rad. And it seems to be working right now but hard to say I haven't got the maps to display yet, I've created page in cfg using model name and tried profile 1 2 3 but no go yet, Its only showing gps coordinates and timer,I'm sure problem is my setup but will keep trying

yaapu, 2021-04-29:
Hi, you should probably stick to "model" for sensors, when you choose "profile 1" the sensor file will no more be model specific and has to be renamed as "profile_1_sensors_maps.lua".

Eg681, 2021-04-29:
Ok thanks for tip I'll keep working on this and will post if I ever get everything working

Eg681, 2021-04-29:
Sorry But I've tried everything I can thank of but can't get maps or sensors to display, I'll send files maybe you can spot the problem, I'm sure it's on my end but don't know were

Eg681, 2021-04-29:
MAPS.zip
<https://drive.google.com/file/d/1R9ieo5PeAIUikE8bta1jYAbaCWLKiGoP/view?usp=drive_web>
Heres my files from sd card, i've tried to set everything up like i've done
with the old script but no joy, I hope you will take alook and maybe see
the problem. On radio screen I do see the yaapu script and showing gps
coordinates and timer but says no map data, in the maps tools section im
using model, default zoom 3, min. zoom 1 and max zoom 8 I have radio left
slider setup on channel 32 for zooming but don't know if it works until I
get maps to display. Thanks

yaapu, 2021-04-29:
download permission denied

Eg681, 2021-04-29:
[CFG.zip](https://github.com/yaapu/HorusMappingWidget/files/6401630/CFG.zip)
[LIB.zip](https://github.com/yaapu/HorusMappingWidget/files/6401633/LIB.zip)
Please try again, I've zipped these files they should open If you need to see my maps file let me know. Thanks

yaapu, 2021-04-29:
Hi, sensor config files should not be in CFG/examples folder but rather in the CFG one along the 

WRONG
[image]

RIGHT
[image]

Eg681, 2021-04-29:
I didn't put that there but can I just copy and paste it over were it belongs, how about my sensors file is it ok? Do you think this is the problem

yaapu, 2021-04-29:
yes that's the problem, the file is fine, it has no errors

Eg681, 2021-04-29:
Thank you, I will make changes and report back, I'm really excited to try this out

Eg681, 2021-04-29:
So I copy and paste the cfg file to correct place and deleted from wrong, I then had to reset everything in the maps tool section still maps or sensors not displaying, after checking were I deleted cfg file it's back again, I keep deleting it keeps appearing in same wrong place. I've used radio and pc to delete but some how it's coming back.

yaapu, 2021-04-29:
you do not have to move the cfg file! keep it where it is, you have to move the lua files out of the CFG/examples folder to the CFG folder like in teh screenshots above

Eg681, 2021-04-29:
[CFG.zip](https://github.com/yaapu/HorusMappingWidget/files/6402104/CFG.zip)
stupid mistake on my part but i did make changes like examples you sent, now still no maps displaying but all sensors now displaying. this is cfg file after changes

Eg681, 2021-04-29:
I do have all 10 sensor's displaying around were map should be but not all is showing a value, without flying to check looks like no cell v, batt. V or fuel no values there

Eg681, 2021-04-30:
So just as update I've added Darren's crossfire sensors file to profile 1 and choose this under yaapu maps. All 10 sensors are now working and showing values, but maps still will not display I can see zoom level changing using left slider but not sure if its actually working because no maps.

Eg681, 2021-05-01:
https://linksharing.samsungcloud.com/h2JHlpAckUy5 this is screen shots showing gps coordinates of my to scripts

MrD-RC, 2021-05-01:
@Eg681 They are both showing the exact same location. They are just using different coordinate formats. You can see the different formats in the search boxes, and the same location in the maps.

[image]
https://www.google.com/maps/place/37%C2%B058'49.8%22N+81%C2%B007'38.1%22W/@37.980511,-81.1294237,17z/data=!4m5!3m4!1s0x0:0x0!8m2!3d37.980511!4d-81.127235

Eg681, 2021-05-01:
That's actually pretty cool, so I'm in process of deleting maps and try to redownload what do you think?

Eg681, 2021-05-01:
https://youtu.be/tGEvquFWit0 maps still won't display only sensors, here's short video of my sd card contents and yaapu map settings. GPS coordinates of downloaded maps have been verified so I don't think problem is in gmapcatcher. Thanks

Eg681, 2021-05-02:
I've done what troubleshooting I can but maps won't display on radio only sensors, this is link for a short video of my setup on radio and sd card.  Any help will be appreciated. https://youtu.be/tGEvquFWit0

Eg681, 2021-05-03:
Just a update maps still not displaying. Darren noticed sat tiles folder was missing from yaapu-script-map so I corrected that see video link. Thanks https://youtu.be/RTWAlwWQgA4

MrD-RC, 2021-05-03:
The full screen required should only come up when the top bar and sliders are enabled. If you page through the telemetry set up screens, you might find they’re enabled for screen 2? You’re only looking at screen 1 here. 

Did you check the maps folders exist? 
If you look in the scripts/yaapu/maps/sat_tiles/2/ folder. You should have a folder numbered 8. Inside there should be folders numbered 806 through to 808. Inside each of those should be a folder numbered 12, which contain the images. If you don’t have those folders, it won’t find your area for level 2 zoom. If they are there, zoom level 2 should work for those coordinates.

Eg681, 2021-05-03:
I don't know why full screen required is coming up when ever radio is booted up, top bars and sliders is not enabled in telemetry page 1 or 2 so maybe something going on with that I also replaced the no map data icon back like it was and I've checked my maps tiles and I don't see a problem there. I must be missing something please take a look at this short video Thanks again https://youtu.be/7cygvgvlq38

MrD-RC, 2021-05-03:
@Eg681 can you record a video showing the folders asked about here https://github.com/yaapu/HorusMappingWidget/issues/27#issuecomment-831038341

Eg681, 2021-05-03:
I'm  not the best a making a video but here it is, it shows telemetry pages 1 and 2 trims sliders not enabled but still getting full screen required message also shows folder 2 in maps looks like images are their. Any help will be appreciated

Eg681, 2021-05-03:
Sorry here's link to video

MrD-RC, 2021-05-03:
Here’s the link ;) https://youtu.be/qoMtGp7Icsw

I’ll take a proper look in the morning and check the tile numbers. 

As for the “full screen required”. Whatever that was, was on telemetry screen 2; as it changed to the INAV LUA. That is actually your home screen. You use the page buttons to switch between the telemetry screens from the home screen. Pressing the _tele_ button is to edit the screens. If you set telemetry screen 1 to be Yaapu maps. When you first switch on, use the _page up/dn_ buttons to change the telemetry screens. You shouldn’t need to use the _tele_ button unless you want to change what is displayed on the telemetry screens.

Eg681, 2021-05-03:
Ok Thanks I'll give that a try on the telemetry pages but as you can see both pages tool bar sliders and trims are not enabled, so I don't understand the full screen required message but I'll give it a try the way you've explained. Thanks again

Eg681, 2021-05-03:
Your right on telemetry pages, now yaapu is home page and I switch pages with up down switch and no more full screen required message

MrD-RC, 2021-05-04:
Actually @Eg681, your homepage will always be the model information, with any telemetry you've set up there. If you keep pressing page up, you'll see what I mean.

As for the maps, you can't see anything because the map data is not there. I have paused the video at the appropriate place in the folder. On the right, you can see the GMapCatcher area for the GPS coordinates, and above that the tiles it needs. They're not on your SD Card.
[image]

If you download and put these on your SD card it should work.
[maps.zip](https://github.com/yaapu/HorusMappingWidget/files/6419445/maps.zip)

yaapu, 2021-05-04:
@MrD-RC Darren you're hired :-)

MrD-RC, 2021-05-04:
@yaapu  😆 I'd never get any time to build my own planes then Alessandro.

Eg681, 2021-05-04:
I've posted before that you are a true professional in this hobby and you confirm this again. After work tonight I'll give this a try and report back. I'm confused as why all the information is not there from when maps were downloaded and the yaapu  widget and sat tiles didn't make it over to sd card in the beginning? Thanks so much I will report my results.

MrD-RC, 2021-05-04:
Thanks Edwin.

Eg681, 2021-05-04:
I'm still struggling, I downloaded the zip you posted,when  to script, yaapu, maps copied to sd card. Still no maps, I tried opening zip file and copied over the sat tiles to sd card but still no go. I'm sure problem is on my end but what am I missing?

MrD-RC, 2021-05-04:
Check the files in the zip file are on the SD Card. The problem is that you are missing the maps.

Eg681, 2021-05-04:
When going to script, yaapu maps what should be there I'm thinking only sat tiles not map folder again?

MrD-RC, 2021-05-04:
Take the sat_tiles folder in the zip as the same as the sat_tiles folder on the SD Card. So in SCRIPTS/YAAPU/MAPS/

Eg681, 2021-05-04:
Ok I've done that, radio showing less than half of map maybe 1/4

Eg681, 2021-05-04:
https://linksharing.samsungcloud.com/iA8eC0icu1WV

MrD-RC, 2021-05-04:
I just downloaded a small area around that location. The fact that the map is showing just means the only reason it didn’t show was because the map data wasn’t there. If you want to see a larger area when zoomed out, you’ll just need to download those tiles with GMapCatcher.

Eg681, 2021-05-04:
-1 and 0 is only zoom levels with full screen, I wonder if it's were I'm using left slider for zoom, I simply went into radio and set it to channel 32 no gv the way you did

Eg681, 2021-05-05:
I deleted and redownload  1 through 7 and it's all working now zoom is very slow but just happy to have maps. I couldn't have done it without all the help thanks again

closed 2021-05-05 by Eg681