yaapu/HorusMappingWidget #29: Generic sensors configs

MrD-RC, 2021-04-06:
I was wondering is if there is any way of creating generic sensor config files?

Currently the sensor config files are modelname_sensors_maps.lua. This is great because it allows flexibility. But in reality, I think that people will only really use different sensor based on the link protocol. 

Would it be possible for the user to create some sensor files, for example:
smartport_gen_sensors_maps.lua
mavlink_gen_sensors_maps.lua

Then, in the configuration of the model’s cfg file, the user can select which sensors file to use.  Or, if a modelname_sensors_maps.lua file exists, that is used and not the file in the cfg. 

I just think, for the most part, it would be quicker to set up maps on new models with generic sensors.

yaapu, 2021-04-18:
Hi it's already that way, the script looks for the model specific one and if not found checks if a file named "default_sensors_maps.lua" exists

MrD-RC, 2021-04-18:
It’s not quite the same. The default_sensors_maps.lua is a fallback of there is not one for a specific model. But it doesn’t know what telemetry you’re using, so the sensors could be completely wrong. 

What I’d like to see is the ability to select a sensor config file, which can be standardised for either SmartPort sensors or Crossfire sensors for example. As both those link have different telemetry data and therefore sensors.

yaapu, 2021-04-18:
ok, I'll add a default 1,2,3,4,5 etc mapping, should be ok

yaapu, 2021-04-18:
[HorusMappingWidget-1.2.2dev.zip](https://github.com/yaapu/HorusMappingWidget/files/6331787/HorusMappingWidget-1.2.2dev.zip)

use the config menu and select the sensor file type: model, profile1, profile2 or profile3
model falls back to default_sensors_map.lua, the other options look for profile_[1,2,3]_sensors_maps.lua

HazzaHFPV, 2021-04-18:
I am sorry to announce that I am unable to get Version 1.2.2 working on my Jumper T16. v1.0 seemed to work fine.
I tried overwriting just the old files with the new as well as deleteing all old folders asnd files and reinstalled the widget including the re-creation of the map tiles. Widget version 1.2.2 however just displays "disabled".
v1.0 works, or at least the widget it displayed..

The content of the "SD" folder in the Archive downloaded via the github release page seems suspicously small compared to v1.0? - If that even means something..

I am no pro. So a user error from my side while installing is quite possible...

MrD-RC, 2021-04-19:
I'm getting the same error on the Horus X10S Express.

My config:
Heading sensor: Hdg
Heading unit: deg
sensor file: _Have tried model and profile 1. Profile 1 was working on my v1.0 installation with the same aircraft model_
Map provider: GMapCatcher
Map type: satellite _I'm using the map data from v1.0_
Map grid lines: no
Map trail dots: 10
Wheel channel: 32 _again, as on v1.0. Also, where is the setting to skip zoom levels?_
Map default zoom level: -1 _map data does exist_
Map min zoom level: -1
Map max zoom level: 3 _I have data for levels -1, 1, and 3_

MrD-RC, 2021-04-19:
> The content of the "SD" folder in the Archive downloaded via the github release page seems suspicously small compared to v1.0? - If that even means something..
> 
> I am no pro. So a user error from my side while installing is quite possible...
> [image]

V1.0 had map data in the SD card folder. That is not there in this version.

MrD-RC, 2021-04-19:
Just tried with Google Maps, and the same result.

yaapu, 2021-04-19:
damn, I have it running on my radio, I must have corrupted the uploaded version, let me check it

yaapu, 2021-04-19:
@MrD-RC I've found the issue, 2 files were missing in the /SCRIPTS/YAAPU/LIB/ folder, sorry :-)

Zoom control is in the lua sensors config file, check at the end the zoomLevels table, you add single levels between min max and the widget will skip those not found there

HazzaHFPV, 2021-04-20:
Thanks for the hard work :)
v1.2.2 working now nicely

MrD-RC, 2021-04-20:
I'm still having the same problem.

I've re-downloaded 1.2.2 and can confirm the 2 luac files (mapsdraw.luac & mapslayout.luac) are in the LIB folder. I've copied everything over from the zip to the transmitter (confirmed to overwrite all). Even deleted all the luac files with a corresponding lua files. Restarted the transmitter multiple times.

I still have **refresh(): ?:0: attempted to index a string value** when I check the _Widget Setting_ on the telemetry setup screen, and attempting to run the widget results in **Disabled**.

yaapu, 2021-04-20:
@MrD-RC you need to clone the latest dev branch, the files were added to the repo not to the release...sorry about that, will fix asap

MrD-RC, 2021-04-21:
@yaapu It's all working now. But I have some feedback for you.
###Zooming
####Switching
At the moment, there is a big delay in the virtual scroll wheel for changing the map scale. I know why. Because it could be used with a regular 3 position switch. However, it would be great if there was a setting in the configuration to reduce or remove the delay. I'm using a repurposed trim switch, so it just needs a click to change the scale.
####Switch direction
On google maps, the zoom in and out of the switch is reversed. For example, I use T6 as my scale switch. With GMapCatcher maps, the channel at 100% zooms in and -100 zooms out. With Google as the map provider, 100% channel values zooms out, and -100% zooms in. I guess this is because the number for the scale is reversed on Google. As in GMap -2 is the most detailed zoom level, but with Google, 20 is the most detailed.
####Min/Max zoom level setting
With google maps, I tried to set the Min zoom level to 19 and the max to 1. Due to the scaling, this would be correct. But, it means you can't actually change the zoom level. Setting min to 1 and max to 19 works fine.
####General
- On the map screen, the text for the model name, RSSI, and Tx voltage hangs down a little, and makes a black box on the map. It's only a couple of pixels, but it would be nicer if the text was all contained within the bar.
- Could the zoom level get a semi-opaque box, like the scale and the coordinates? This would make it easier to see.
###Setup script
Changing the values seems backwards here. Rotating the scroll wheel clockwise decreases values.and anti-clockwise increases. This seems backwards to me, and trips me up every time I change a value.

I've not tested with Crossfire yet. But on SmartPort, it's looking good. Thanks your work Alessandro. I can post screen grabs or example videos in the evening, if they'll help.

Error404fpv, 2021-04-22:
I use t16pro Radio/google map/ Zooming
As soon as I use the telemetry file for the model, the Zooming map stops working. It stands in the main position that is selected in the settings.

closed 2021-04-29 by yaapu