yaapu/HorusMappingWidget #30: Jumper T16 - settings menu problem

jaroszmm, 2021-04-08:
This is going to be a long post but I've noticed a weird issue with the Widget.

I am on Jumper T16 with the newest opentx 2.3.11.
I followed two youtube films (Marc's and Mr D's)
https://www.youtube.com/watch?v=LYsgsCoBK34
https://www.youtube.com/watch?v=KIibX_zK2VU

and faced a weird issue. I cannot execute the settings for the maps. If I try like on the videos and go from the main tools screen into  the Yappu Maps and enable it I get a different settings page, with a lot of more parameters. It seems to be the settings menu for the Ardupilot version of the Yappu Maps.
At this point I checked if the correct files were downloaded, cleared the SD card etc but no luck. I still get that menu in the Tools page. And this menu doesn't do anything, doesn't control the maps widget.

HOWEVER...
If I go SYSTEM - SD CARD - SCRIPTS - YAAPU - mapsconfig.lua and execute this script I am able to use it to set up the widget - enable or disable grid, change zoom channel and zoom level.

BUT....
This works ONLY if I use the SD card content from Github marked sdcard-480x272-2.3V0034.zip 
from this
https://downloads.open-tx.org/2.3/release/sdcard/opentx-t16/ 
folder. 
The trick is to use this content, edit the firmware file on the SD card from 0034 to 0035 so my transmitter doesn't get the error of wrong firmware on SD card expected.

If I on the other hand use the SD card file sdcard-480x272-2.3V0035.zip  then when I try to execute the mapsconfig.lua script from the same location SYSTEM - SD CARD - SCRIPTS - YAAPU - mapsconfig.lua I get a brief flash of the menu but immediatelly after I get an error message "Bad argument #3 to drawText" and the script doesn't execute.

CONCLUSION:
As I was starting from scratch with a fully formatted SD card it seems like on my Jumper T16 the newest SD card contents somehow create a conflict with the Yappu Widget or part of it. Yet it seems to be limited to Jumper T16 at least because Marc and Mr D use other transmitters.

b14ckyy, 2021-04-10:
Saw the same issue reported by someone with a Radiomaster TX16S.
I wonder if it has something to do with, that I initialized the Radio with the Factory SD Card Content first and then just Updated the SD Card Content from Radiomaster. It is really hard to narrow down what causes the issue.

jaroszmm, 2021-04-11:
My Jumper is a second hand one so can't say how it was configured at first.

b14ckyy, 2021-04-11:
@jaroszmm The Horus-Type radios have no internal configuration storage. All data are stored on the SD card. So if you replace the SD card, this is a complete factory reset. Even calibration data. 

Anyway let's hope we figure that out soon. More people start using it now.

yaapu, 2021-04-18:
Hi, I suspect this has nothing to do with the widget itself.
I have an Horus X10, a Radiomaster TX16S and a Jumper T16 and tested my widget against all of them with OpenTX up to 2.3.11, provided you match sd card content with OpenTX version I never had an issue, not sure what's going on in your specific case.
OpenTX 2.3.x has minimal changes between Horus class radios so it's very unlikely that something works on only one of them.

closed 2021-04-18 by yaapu