yaapu/HorusMappingWidget #31: No telemetry data - just gps coordinates

jaroszmm, 2021-04-21:
One more issue about my yaapu on Jumper T16

I get no data on my telemetry screen, meaning I see the arrow moving on the map, I get the GPS coordinates but that's it. No current, no voltage, nothing. 
I have LUA Telemetry script too on a second screen and the data there is present.
Any ideas? 
I might upgrade my Crossfire receivers and tx module to the newest software and see if it makes any difference.

Eg681, 2021-04-22:
I can confirm any crossfire firmware below 4.8.0 will cause yaapu and inav telemetry issues, in my case radio wrong audio flight modes, after updating to 4.11 that problem now gone

yaapu, 2021-04-25:
@jaroszmm telemetry data needs a lua configuration file specific to your setup, did you create the file and customize it?
There's no widget magic that figures telemetry data for you sorry, you need to provide a valid lua config file

jaroszmm, 2021-04-26:
OK, I will try it again - use the sample file, rename it according to the model name and this should make it work, right?

jaroszmm, 2021-04-27:
UPDATE:
Newest crossfire software doesn't help. I still get zeros on all values except for gps data.

yaapu, 2021-04-29:
I'm pretty sure it's an issue on your side, most probably the file name is not correct or content was corrupted, please attach your cfg file and your sensors_maps file

jaroszmm, 2021-04-29:
Darren Lines sent me his config file for crossfire system. I renamed the file to fit my model name and it worked.
If I did the same thing with the file that comes with the script (there are 3 sample files as far as I remember) I got no results.
So yes, something on my side, 99% user error.
Oh, and there is still one funny thing:
If I go to the configuration menu from SYS page I get a weird menu (probably for ardupilot) with a LOT of options.
But if I navigate to the SD card / Scripts / Yaapu etc and use the config menu there everything works.
But I had to use on my Jumper T16 SD Card contents from one but last opentx version.
Weird. 
But it works.

Eg681, 2021-04-29:
I'm also trying to use Darren's crossfire sensors file and renamed to model but still no maps or sensors displaying, I'm not on the latest opentx but running 2.3.9 which was working with previous yaapu version.

yaapu, 2021-04-29:
> Darren Lines sent me his config file for crossfire system. I renamed the file to fit my model name and it worked.
> If I did the same thing with the file that comes with the script (there are 3 sample [...]

The right config tool is "Yaapu Maps", "Yaapu Config" is for the ardupilot widget which I have no idea how you got on your SD card :-)

The sensor file is just a matter of setting the correct sensor names, perhaps change the one that worsk with your sensor names.
Sensor support is a bit tricky and requires some patience

jaroszmm, 2021-05-01:
I have no idea how it appeared on my transmitter. I used the SD card content that comes from OpenTX folder, fully formatted SD card and then copied the files as the manual says to enable yaapu telemetry. And somehow I got the ardupilot too. 
Anyway - it works :) It's nice to see the plane on a minimap :)

wolfeeldiver, 2021-05-28:
Hello. I am having  NO MAP DATA tiles being shown on my Radiomaster TX16S controller.
I have verified that map tiles were downloaded, using 0/8.2.1, and were saved to a 100x100 folder.
WHen downloading my area, I selected -2 to 0 
Those subfolders in the 100x100 folder copied to my radio's MSD card to the following :
\scripts\yaapu\maps\sat_tiles\0\*
I have confirmed that on the radio's sd card are \0   \-1 and \-2 folder, and each do contain many sub folder, and eventually .PNG images. The images are sized 100x100 and are of the area that I downloaded, and I have confimed am very confident that the map tiles are on the controllers SD card.
I can confirm that the GPS coordinates telemetry is being passed to the mapping widget. 
The quad is powered up, it sees sats, I see the GPS coordinates displayed in the Mapping widget. 
I have confirmed that the coordinates displayed in the widget is correct, and the location is within the area where I downloaded tiles.
I looked at the controller's TOOLS, Yaapu Maps, tool, and that is configured as follows:
map zoom level -2
Map type   Satellite
Grid Lines Yes
TRail dots 10
Maps zoom channel  is --
I am using Yaapu Mapping Widget 1.1-dev it says.

Any suggestions are greatly appreciate to make the map tiles actually show up. 
Thank you.

yaapu, 2026-04-23:
please reopen if needed, perhaps you solved the configuration issue by yourself :-)

closed 2026-04-23 by yaapu