yaapu/HorusMappingWidget #32: APM Planner 2 on Linux

dentex, 2021-04-22:
Hi,
I installed the Mission Planner in order to download the maps tiles, but I had a few issues. 
This new v2 downloads the files into `/home/$USER/apmplanner2/mapscache/Data.qmdb`.
I run the resize tool in wine and I was not able to let it point to this dir because it exposes some hard-coded paths mixing windows and Linux path separators (images for reference). How to proceed? Thanks.

[image]
[image]

yaapu, 2021-04-25:
Hi, that's APM Planner, I only support mission planner, different product.
As for running the resize tool on linux, that's another story, need to experiment with it, for now I'd suggest copying the downloaded tiles to a wine reachable path

dentex, 2021-04-25:
Oh OK, I see now they are two different products. Unfortunately Mission Planner is a windows only application and under wine gives too many errors and it doesn't even install. So I'm out. 
Sorry.

closed 2021-04-25 by dentex