yaapu/HorusMappingWidget #33: Mapping Widget version 1.2.2b Feedback

MrD-RC, 2021-04-24:
Hi @yaapu It's all working now. But I have some feedback for you. I did originally have this in #29, but I think it would be better in it's own issue. This way others can add feedback, and it's easier to find.

### Zooming
#### Switching
At the moment, there is a big delay in the virtual scroll wheel for changing the map scale. I know why. Because it could be used with a regular 3 position switch. However, it would be great if there was a setting in the configuration to reduce or remove the delay. I'm using a repurposed trim switch, so it just needs a click to change the scale.
#### Switch direction
On google maps, the zoom in and out of the switch is reversed. For example, I use T6 as my scale switch. With GMapCatcher maps, the channel at 100% zooms in and -100 zooms out. With Google as the map provider, 100% channel values zooms out, and -100% zooms in. I guess this is because the number for the scale is reversed on Google. As in GMap -2 is the most detailed zoom level, but with Google, 20 is the most detailed.
#### Min/Max zoom level setting
With google maps, I tried to set the Min zoom level to 19 and the max to 1. Due to the scaling, this would be correct. But, it means you can't actually change the zoom level. Setting min to 1 and max to 19 works fine.
#### General
- On the map screen, the text for the model name, RSSI, and Tx voltage hangs down a little, and makes a black box on the map. It's only a couple of pixels, but it would be nicer if the text was all contained within the bar.
- Could the zoom level get a semi-opaque box, like the scale and the coordinates? This would make it easier to see.
### Setup script
Changing the values seems backwards here. Rotating the scroll wheel clockwise decreases values.and anti-clockwise increases. This seems backwards to me, and trips me up every time I change a value.

I've not tested with Crossfire yet. But on SmartPort, it's looking good. Thanks your work Alessandro. I can post screen grabs or example videos in the evening, if they'll help.

MrD-RC, 2021-04-24:
One other thing I've noticed. Would it be possible to add support for Bing maps here. I've noticed in Mission Planner that quite a few of the Google maps tiles are missing due to an over use of resources. The Bing maps seem to download faster too.

[image]

Is this just a case of folder names?

yaapu, 2021-04-25:
Hi Darren, Bing maps is supported thru gmapcacther, the "Virtualearth" provider is bing

[image]

MrD-RC, 2021-04-25:
True @yaapu, I admit I forgot about that. However, I'm wondering if it might be better to use this new method, as the keeper of GMapCatcher doesn't seem keen on maintaining it anymore. Maybe this is something that could be considered for the future? On a similar note, I have asked for some enhancements to the Mission Planner prefetch (_min zoom_ in addition to the current _max zoom_ and an option to ignore already pre-fetched tiles).

What are your thoughts on my other suggestions?

yaapu, 2021-04-25:
@MrD-RC I need to double check MissionPlanner code, I suspect Bing has a different tile engine, on gmapcatcher the tile engine was unified, it was the orignal author who did the abstraction between the source tile system and gmapcacther's one, in mission planner the auther used google tile system by the mean of a custom library, I suspect bing has it's own which would mean adding another tile system to the widget which might end up in a rather complex project.

I mostly agree on your other suggestions, I will take a look and address them, I like the idea of quick zoom change

MrD-RC, 2021-04-25:
@yaapu to me, it looks like they both use the same numbering systems. I just thought I'd have a look at 2 downloaded folders. The scales appear to be the same too.

[image]

Looking at this, I think if I were to just rename the BingSatelliteMap folder to GoogleSatelliteMap, it would work. Overall, I think the Google maps look nicer. It's just frustrating that there are huge areas missing.

Cool. Overall, I think these are great changes, to an already great widget. I'll definitely be making a new video on this version once it is released as a stable version.

MrD-RC, 2021-04-27:
Works great on Crossfire.

yaapu, 2021-04-29:
How do you repurpose the trim switch and how do you actually use it for zoom?

yaapu, 2021-04-29:
new zoom handling:
zoom change triggers right away and user can select how long to wait for next level triggering from the config menu

[HorusMappingWidget-1.2dev.zip](https://github.com/yaapu/HorusMappingWidget/files/6401442/HorusMappingWidget-1.2dev.zip)

As a side note I can't see the overflowing pixels from the top bar in the simulator, could it be a radio model specific issue?
Min max zom levels both in the config menu and on screen actually refer to zoom level "values" which as you have noted are reversed in gmapcatcher vs google, but my code already handles that.
I can reverse the behaviour of the switches for right now -100 increases zoom

MrD-RC, 2021-04-29:
> How do you repurpose the trim switch and how do you actually use it for zoom?

- First I disassociate the trim in the flight modes. So trim 6 for example shows —
- Next I’ll create the mix that’s used to change the zoom level. I use channel 32 as that’s out the way. There are 3 lines to the mix:
1. Source: Max | Weight: 0%
2. Source: Max | Weight: 100% | Switch: trim switch 6 up
3. Source: Max | Weight: -100% | Switch: trim switch 6 down
- Then just set the zoom channel to 32 in the Yaapu maps config tool.

MrD-RC, 2021-04-29:
Cool, I’ll check out the new version on Saturday. 

The overflowing pixels is on the X10S Express. I can get a screen grab at the weekend too.

yaapu, 2021-04-29:
> * First I disassociate the trim in the flight modes. So trim 6 for example shows —
> * Next I’ll create the mix that’s used to change the zoom level. I use channel 32 as that’s out the way. There are [...]

...so you use it as a temporary 3 pos switch

MrD-RC, 2021-04-29:
Yes, it effectively turns the trim switch in to a bidirectional momentary switch.

yaapu, 2021-04-29:
Minor update with zoom overlay and faster zoom change detection

[HorusMappingWidget-1.2dev.zip](https://github.com/yaapu/HorusMappingWidget/files/6401877/HorusMappingWidget-1.2dev.zip)

MrD-RC, 2021-05-01:
I've just installed the new dev release. I've not tested with a GPS connected yet, but thought I'd get a screen grab of the hanging boxes. The semi-opaque box around the zoom level looks great in your screen shot.

Here's the pixel issue on the X10S Express.
[image]

yaapu, 2021-05-04:
@MrD-RC Darren, I really can't reproduce the top bar issue, I tried on a Jumper TY16, radiomaster tX16S and Horus X10 with OpenTX 2.3.11 and all work, I also tried Horux X10S Express in the simulator and the layout is fine... did you do anything special on your radio, did you change fonts, is it a custom build?

HazzaHFPV, 2021-05-04:
With the latest version provided here and on the release page, I noticed the following:
when I try to use "Google" as Map Provider and "GoogleSatelliteMap" data downloaded via MissionPlanner I cannot get the zoom to work correctly. Zooming in does nothing and zooming out jumps immediately to level 3, without being able to zoom back in again.
The map data is all there, if I select any zoom level as default it will display the map correctly (levels 1-20) upon restarting the radio, but as soon as I zoom out it will jump to level 3. Zooming in is not possible, no matter at which level I start as default.

I am hesitating to download all data again via GMapCatcher as it took me several hours to get all the zoom levels downloaded for my area :/

When I chose GMapCatcher as map provider it switches zoom levels without problem.

HazzaHFPV, 2021-05-05:
So after hours of trying I can now claim the following:
Zoom works as long as I leave [modelname]_sensors_maps.lua as completely empty file. As soon as I configure some sensors within that lua, which then are correctly displayed within the widget, zooming does not function anymore. Same happens when using the profile 1-3 lua files. If I select profile1, for example, but delete that lua file = no sensors showm (of course) and zooming works. As soon as I have sensors shown, zooming in/out does not work anymore.
Radio is a Jumper T16 with OpenTX 2.3.11

yaapu, 2022-11-27:
stale issue, please reopen if not fixed by latest version, thnaks

closed 2022-11-27 by yaapu