yaapu/HorusMappingWidget #36: Beta version. Cant Zoom map with active telemetry Lua file being displayed on X12S

FLYN-001, 2021-05-18:
I am unable to zoom the map on my X12S with CRSF when a profile is selected that has a telemetry Lua & is being run to display data on the map screen, even your own example files from the download, Im using google maps hybrid satellite maps.

If I select a profile that has no telemetry Lua file present on the SD card no telemetry data is displayed & the map zooms perfectly normally. But if I flick over to a profile that has a telemetry Lua file to display any info on the screen the map stops zooming. 

Maybe I did something wrong, maybe there's an issue. Happy to help you get to the bottom of it. Thanks

HazzaHFPV, 2021-05-18:
I fell into a similar problem until I noticed that it is necessary to define the valid zoom levels that you would like to use within the profile Lua which you select. At the bottom of the Lua, right at the end, is a section for defining the zoom levels.
My zooming did not work either until I entered valid zoom levels in that section.

FLYN-001, 2021-05-19:
> I fell into a similar problem until I noticed that it is necessary to define the valid zoom levels that you would like to use within the profile Lua which you select. At the bottom of the Lua, right at the end, is a section for defining the zoom levels.
> My zooming did not work either until I entered valid zoom levels in that section.
> 
> Edit: see attached screenshot
> [image]

Thank you so much, it was very late last night I was installing things, totally didn't see this! Thanks so much for you reply. Issue sorted, zoom working!

closed 2021-05-19 by FLYN-001

HazzaHFPV, 2021-05-19:
Yep, same happened to me. It was 2am when I tried to set everything up and it took me an embarrassing long time to spot this :) glad I'm not alone though :D

FLYN-001, 2021-05-19:
Yeah exactly - sometimes you just gotta call it a night & look again with fresh eyes in the morning, because its just so easy not to see stuff that's written right there!! haha