yaapu/HorusMappingWidget #37: Have GPS data - but NO MAP DATA being shown on screen.

wolfeeldiver, 2021-06-05:
Hello. I am having NO MAP DATA tiles being shown on my Radiomaster TX16S controller.
I have verified that map tiles were downloaded, using the modified GmapCatcher 0.8.2.1, and were saved to a 100x100 folder.
WHen downloading my map area, I selected -2 to 0
Only the subfolders in the 100x100 folder were copied to my radio's MSD card to the following folder:
\scripts\yaapu\maps\sat_tiles\0*
I have copied only the \0, -1 and -2 folders to the controller SD card.
I have confirmed that on the radio's sd card \0 -1 and -2 folder and all their sub folders and files were copied to the SD card. 
I verified that the images are sized 100x100 and are of the area that I downloaded, and I am very confident that the map tiles are on the controllers SD card.
I can confirm that the GPS coordinates telemetry is being passed to the mapping widget.
The quad is powered up, it sees the sats, I see the GPS coordinates and is displayed in the Mapping widget lower right of where the 8 tiles are displayed on the screen. Except those tiles simply say NO MAP DATA.
I have confirmed that the coordinates displayed in the widget are correct, and that the location is within the area where I downloaded tiles.
I reviewed the controller's TOOLS, Yaapu Maps, tool, and that is configured as follows:
map zoom level -2
Map type Satellite
Grid Lines Yes
TRail dots 10
Maps zoom channel is --
I am using Yaapu Mapping Widget 1.1-dev it says.

Any suggestions are greatly appreciate to make the map tiles actually show up.
Thank you.

ramiss, 2021-06-23:
Did you ever solve this?  I'm having the exact same issue.

allibongo, 2021-08-29:
yes,installed version 1.0 instead if version 1.2.2.

audiogek, 2021-11-03:
Using 1.2.2dev and Missionplanner to download maps. 
Maps are in a folder named "GoogleSatelliteMap" with 20 as max zoom level and 1 zoomed out max.
Had some trouble too and found a different gmapresizetool that makes that folder structure. After that it worked for me.
Sadly I don't remember where I found it and can't find the page.

wcfung, 2022-01-12:
Hi,
I  used Mission planner's Map tool  to prefetch, but could not find the prefetched map in C:\ Program files(x86) \mission planner \gmapcache. Do you know the possible reason ?   
Also I searched all over the internet and cannot find the gmapresizetool.zip mentioned in Github. Pls direct me to the link.
Thank you.

audiogek, 2022-01-13:
Check here: C:\Users\*username*\.GMapCatcher
or here: C:\ProgramData\Mission Planner\gmapcache\TileDBv3\en\GoogleSatelliteMap

And this is what I have. A combination of the two worked for me:
https://drive.google.com/file/d/1Nnbd-8krYBDA1SOWymiQBvp__lbXOFOA/view?usp=sharing
https://drive.google.com/file/d/1dkvZkv-duA3RBa1knihFb9oNnccTvPxD/view?usp=sharing (Look in tools folder for gmapresizetool)

On the transmitter: \SCRIPTS\YAAPU\MAPS\GoogleSatelliteMap\ and then folders 1-20 (zoom lvl 20=max. zoom lvl 1= whole earth)
But one of the widgets wants this: \SCRIPTS\YAAPU\MAPS\sat_tiles\ and then folders 1, 0, -1 

Good luck!

wcfung, 2022-03-01:
> I have the same issue. Getting bald pulling my hair.
> I have telemetry data on the right and bottom. I even have the GPS
> coordinates. The only thing missing is the map. I'm on 1.2.2b .

Did you get the problem solved ?

Attached is the display of my Jumper T16 on the mapping widget. Sometimes a
tile of map shows up
in one of the grids. I don't know why not on other grid.

Let me know if you have the problem solved.

Tks
Frankie

wcfung, 2022-03-02:
Hi Bookie,

Further to my last reply with mapping widget display on my Jumper T16, I
here provide the settings in the Yaapu Config and Yaapu Maps for
comparison, as follows  :--

*Tools > Yaapu Config (Yaapu Telemetry Widget 1.9.5)* :--
                map provider : Google
                map type :  GoogleHybridMap
                map zoom level default value  :  1
                map zoom level min value      :  1
                map zoom level max value     :  12

*Tools >Yaapu Maps > Yaapu Mapping Widget 1.3.0* :--
                map provider :  GMapCatcher
                emulated wheel channel* :*   ---
                map provider* :*   Google
                map type *:*        GoogleHybridMap
                map zoom level default value  :  - 2
                map zoom level min value      :  - 2
                map zoom level max value     :  17

Can you advise your setting so I can compare  ?

Thank you.
Frankie

bokie1, 2022-03-02:
> Hi Bookie, Further to my last reply with mapping widget display on my Jumper T16, I here provide the settings in the Yaapu Config and Yaapu Maps for comparison, as follows :-- *Tools > Yaapu Config [...]

Where did you get 1.30?

wcfung, 2022-03-02:
You get it here :--

https://github.com/yaapu/HorusMappingWidget

Can you share your setting in the Tx ? Is yours an Open Tx  transmitter?

bokie1, 2022-03-02:
> *Tools > Yaapu Config (Yaapu Telemetry Widget 1.9.5)* :-- map provider : Google map type : GoogleHybridMap map zoom level default value : 1 map zoom level min value : 1 map zoom level max value : 12 [...]

EdgeTX 2.7 nightly or Edgetx 2.6

emulated wheel channel* :*   16
map provider* :*   Google
map type *:*        GoogleHybridMap or GoogleSatelliteMap
map zoom level default value  :  15
map zoom level min value      :  1
map zoom level max value     :  20

bokie1, 2022-03-06:
It's now working for me.

closed 2022-12-31 by yaapu