yaapu/HorusMappingWidget #41: Map boarder sensors are not displaying

PaulPettman, 2021-11-26:
Hi,

Please help, i have installed everything, i have the map displaying and working correctly, i have not been able to get the map boarder sensors to display, i am using open tx in a radio master TX16S with a happymodel elrs module installed, im getting telemetry info to the receiver ok.. in the TX, when i go to scripts/yaapu/cfg and click on arwinglrs_sensors_maps.lua and select execute the writing go's dim and does not seem to execute like other files do..... i have tried to reinstall everything but just cant seem to crack it....
Please please help..

Thanks Paul

PaulPettman, 2021-11-28:
I have fixed it!!!!!!! yay.... The zip i downloaded did not have the files "profile 1, profile 2 and profile 3" in the CFG/examples folder, i found another zip that had them in, moved them to the CFG folder and boom.......

yaapu, 2022-02-03:
gerat sorry for the very late reply :-(

closed 2022-02-03 by yaapu