yaapu/HorusMappingWidget #42: Maps config tool fields not visible in EdgeTX 2.6

ulfhedlund, 2022-01-05:
I had the mapping widget working in ETX 2.5 but have run into some problems with the configuration tool in ETX 2.6 RC2.
All field info/headers are non-readable/completely black in the config menu. The only thing I can see is the active field value, which is shown in white text on black background.

yaapu, 2022-01-13:
another EdgeTX lua "feature" :-)

ulfhedlund, 2022-01-13:
It seems to also effect the map widget itself and your other excellent FrSky Telemetry Script and Widget, unfortunately.

yaapu, 2022-01-14:
can you provide a screen shot, I tested under the simulator and works

ulfhedlund, 2022-01-14:
Absolutely, I'll provide them to you over this weekend.

ulfhedlund, 2022-01-14:
[tests.zip](https://github.com/yaapu/HorusMappingWidget/files/7873564/tests.zip)
Here are some screenshots from my TX16S, running ETX 2.6 RC3, the FrSky Telemetry Script 1.9.5 and the Horus Mapping widget v1.2.2 (I believe). It seems all the telemetry/text fields along the edge of the widgets are black on black or something like that.
Please let me know if you need more or other info.
Thanks!

yaapu, 2022-01-14:
ok, check my wiki, you need version 1.9.6 of the Telemetry Widget and version 1.3 of the mapping widget for EdgeTX support, use the dev branch

ulfhedlund, 2022-01-15:
Thank you! Everything now works as expected on the radio, both the Telemetry Widget and the Mapping Widget and config scripts. However the mapping widget throws an error in the model simulator in Companion. "ERROR in create(): /SCRIPTS/YAAPU/mapsconfig.luac: incompatible precompiled chunk".

An observation: the background colors of the telemetry values in the widgets are different for those on the side (blue in my case) and those at the bottom (black in my case). It is cosmetics only so no big deal, and perhaps it is intentional.

I have attached a couple of screenshots from companion and radio, showing this.
[tests.zip](https://github.com/yaapu/HorusMappingWidget/files/7874802/tests.zip)

Ziege-One, 2022-02-02:
I have the same problem. the issue is that the mapsconfig.lua will be not compiled at start up. I have added a compiled in Main.lua

` local function init()

-- load configuration at boot and only refresh if GV(8,8) = 1
  assert(loadScript(basePath.."mapsconfig.lua","c"))    --new
  collectgarbage()                                                     --new
  collectgarbage()                                                     --new
  loadConfig()
  utils.initColors()
  -- zoom initialize`

yaapu, 2022-02-02:
that should not be required, simply remove the .luac file and the lua will be compiled on the fly

Ziege-One, 2022-02-02:
I have edgeTX 2.6 it will not compile. A feature for edgeTX ? 😁

yaapu, 2022-02-02:
did you delete the mapsconfig.luac file?

Ziege-One, 2022-02-02:
yes then restart with the original Main.lua mapsconfig.luac not compiled.

yaapu, 2022-02-02:
ok now I see what you mean, you can also manually launch mapsconfig.lua from SYS->[browse the sd],  it will compile it once and for all

Ziege-One, 2022-02-02:
Yes that works 👍

yaapu, 2022-02-02:
but it's an ugly workaround, the "feature" is that luac compiled for OpenTX 2.3 does not run on edgetx

Ziege-One, 2022-02-02:
Another Question, I have made a Fullscreen Map option. Should I make a Pull requests, or should I stay a separate project?

[image]

yaapu, 2022-02-02:
mhmm that won't work on opentx, it would be killed for cpu limit, probably on edgetx too, you're really close to max cpu

yaapu, 2022-02-02:
the proper way would be to add a second layout, user selectable from the config menu and in that layout go full screen, so users would be free to choose and if the radio kills the script would not be my fault :-)

Ziege-One, 2022-02-02:
Where can I see the CPU limit?

yaapu, 2022-02-02:
you can't :-) the radio will kill the script without notice, I do simulated multi hours mission to detect such cases

yaapu, 2022-02-02:
you can try to decreae the number of trailing dots if it gets killed, they are quite cpu heavy

julianschweizer, 2022-02-03:
> ok now I see what you mean, you can also manually launch mapsconfig.lua from SYS->[browse the sd], it will compile it once and for all

Doesnt work here on latest EdgeTX and TX16S

closed 2022-11-27 by yaapu