yaapu/HorusMappingWidget #45: No map screen or visible config on Radiomaster TX16s

Flierman1945, 2022-04-27:
I know there has been a problem with the mapping config screen not showing properly using Edge 2.6 and have tried the fix for this, but my question is whether there is another issue wit the new release of Edge 2.7.  I can get the yappu telemetry screens to work fine but the mapping addon will not work at all for me.  are you aware of any issues with the 2.7 release?. I am using the  link you posted here   https://github.com/yaapu/HorusMappingWidget/archive/refs/heads/master.zip and I believe it is the correct version to get mapping going.

pfeerick, 2022-05-01:
The head version linked of this widget is working for me - both the widget and the config tool, with 2.7.0 of ETX.

Flierman1945, 2022-05-02:
Thanks for your reply. I have managed to get it going now.  Turns out I missed deleting an old yaapu file .  Should have read the full instructions about removing all old yaapu files before writing in the current files that work.  "Before install, delete all older yaapu files."

closed 2022-11-27 by yaapu