yaapu/HorusMappingWidget #48: No Map Data

ammikey, 2022-12-19:
Radio master TX16s running Edge TX 2.8, latest Gmapcatcher, Horus mapping widget 2.0.0b2, Inav 5.1.0.
Yaapu widget and all telemetry working correctly the maps are at 100x100 and on the sd card at ( Images/ Yaapu/Maps/Sat_tiles.
Down load zooms are 2 to 6 and levels are set accordingly in the LUA. I've tried everything I could find online to resolve the problem with no luck. Any help will be greatly appreciated.
Thanks 
Mike Johnson

lbellic, 2022-12-29:
i have the same problem , same radio

yaapu, 2022-12-30:
Hi this is most likely a setup issue.
To double check the obvious:
- make sure you discovered your GPS sensor
- remove zoom level restrictions from the lua sensor config (you can add them later on)
- make sure the widget is setup to use gmapcacther, set min zoom = -2, max zoom=17, default zoom = -2
- setup the zoom control switch as per wiki instruction, set a high delay like 2 or 3 secs

lbellic, 2023-01-02:
> Hi this is most likely a setup issue. To double check the obvious:
>
> * make sure you discovered your GPS sensor
> * remove zoom level restrictions from the lua sensor config (you can add them later on) [...]

i tried this setup , but i can't view the map... in the past i tried the first version of the script and working fine wit 100x100 tiles gmapcatcher, i don't know why this one can't load the maps..

lbellic, 2023-01-02:
i solved the problem by resize every image with mktiles

yaapu, 2023-01-02:
> i solved the problem by resize every image with mktiles

no need to use mktiles! please read the wiki, the gmapcacthr version I provide does resize the tiles down to 100x100

lbellic, 2023-01-02:
ah ok , i figure out that the problem was the path in the /IMAGE folder and not the old path..
not about the size of tile.. Thx script running well , only thing it's buggy is the travel distance

yaapu, 2023-01-02:
> ah ok , i figure out that the problem was the path in the /IMAGE folder and not the old path.. not about the size of tile.. Thx script running well , only thing it's buggy is the travel distance
what do you man buggy?

lbellic, 2023-01-02:
i mean there's bug in the value "distance traveled"

yaapu, 2023-01-02:
ok, but what's the bug? is lower/higher than expected, it's blank, ?? provide a screenshot please

ammikey, 2023-01-05:
Hi, I found the problem is in the file structure on the SD card. The file structure that works is as follows.
[image]

yaapu, 2023-01-05:
@ammikey 
Hi that is not a bug, it depends on the type of tiles you download:
satellite type images path is /sat_tiles, and in the widget you need to select "satellite"
vector type images path is /tiles and in the widget you need to select "map"

so the bug might be in the docs if this is not clear :-)

lwan288, 2023-09-29:
> Hi this is most likely a setup issue. To double check the obvious:
>
> * make sure you discovered your GPS sensor
> * remove zoom level restrictions from the lua sensor config (you can add them later on) [...]

Hi,  I am sorry that I didn't find the lua sensor config. Is the following map lua file?   Thanks!

[image]

[image]