yaapu/HorusMappingWidget #52: Wd love map-centred?

MortyFleem, 2023-02-22:
Hi - love this application. Are there any plans to have the option to have the map centre on the aircraft's location?
If confused: I mean like as in Google maps for example. It keeps (you) in the middle of the screen and the map moves around as you move. Thanks.

fa-lk, 2024-05-24:
nice idea!

FearResonans, 2026-08-29:
Not exactly what was originally asked here, but I found something related while digging into this.

The vehicle icon is never centered on my TX16S MK2 (EdgeTX, 480x272, sidebar and bottom bar disabled, so layout.draw calls drawMap with cols=5, rows=3).

In lib/maplib.lua, loadAndCenterTiles() uses:

lua
local xx = math.floor(width / 2)
local yy = math.floor(height / 2)

With a 5x3 grid this gives xx = 2 and yy = 1, so the vehicle tile lands in column 2 of 5 and row 1 of 3 — the topmost row. Combined with getScreenCoordinates() the icon ends up at screenY = 20..120, while the visible map area centre is around 130, so it always sits near the top edge and slightly left of centre.

Changing those two lines to math.ceil gives xx = 3, yy = 2 (the true centre cell) and the icon sits properly in the middle. Tested on 2.2.0, works fine, no side effects noticed so far.

Interestingly, loadAndCenterTilesFixed() right above already uses math.ceil, and drawTilesFixed() computes the origin from LCD_W/2 / LCD_H/2 — but neither of them is called from drawMap(). Looks like the fix is already written but not wired up yet.

Great widget by the way, thanks for all the work on it.