yaapu/HorusMappingWidget #57: Attempting configuration using only MacOS

kessmess, 2023-11-29:
I have the Yaapu widget installed and working on TX16S, no problem. GPS coordinates detected, all good there. The example GoogleSatelliteMap examples provided show the Australia tiles at zoom level 1. Configuration of yaapu maps on the radio appears be complete. 

I have Mission Planner running on MacOS 10.15 no problem, and have the raw prefetched google satellite tiles downloaded (256 x 256 default size)

I built a simple Automator script to get the 256x256 tiles, scale them to 100x100, and convert them to .PNG and replace them in the same directory.

When I upload the converted tiles (preserving the 1-20 folder structure of course) into /IMAGES/yaapu/maps, only image displayed is nomap.png.

Is there some sort of geo metadata etc. being adding by gmapresizetool.exe that I am missing by simply scaling and converting prefetched tiles? If so can I inject them without the tool?

kessmess, 2023-12-01:
I have answered my own question after studying the output of tiles processed by gmapresizetool.exe.

The steps for creating tiles compatible with Yaapu Mapping Widget with MacOS:

1. Download [Mission Planner for OSX](https://github.com/ArduPilot/MissionPlanner/suites/16375288014/artifacts/935679479). (The version linked here may be old, you may want to find a more recent version).
2. Follow the steps in the wiki for [Google Maps tile download prefetching](https://github.com/yaapu/HorusMappingWidget/wiki/Google-Maps-download-and-configuration#tiles-downloading). The images are exported to home/.local/share/Mission Planner/gmapcache/TileDBv3/en/GoogleSatelliteMap (or GoogleHybridMap etc., depending on your preference). Stop at Tiles resizing.
3. For tile resizing in MacOS use the attached Automator workflow titled 'googlesatellitemap100x100'. In the first action, which is Get Specified Finder Items, click add and select the GoogleSatelliteMap folder containing the raw images from Mission Planner. The attached Automator workflow will resize the images to 100x100 and add the 's_' prefix into each image name. Beware that this workflow will replace the original images with the resized and renamed images. 
4. Copy the images onto the radio's SD card in the correct directory. For Horus Widget version 2.0.0 or later use /IMAGES/yaapu/maps

I was previously missing the 's_' image name prefix in the workflow, which will result in the images failing to load.

kessmess, 2023-12-01:
@yaapu you may want to add this to the wiki :)

closed 2023-12-01 by kessmess

yaapu, 2026-05-08:
Hi, try this new tool, pretty cool and easy!
https://martinovem.github.io/High-Resolution-Map-Generator