yaapu/HorusMappingWidget #62: Flight Modes in the telemetry widget?

rickgode, 2024-12-24:
I was customizing the telemetry sensors on the screen and tried to put the Flight Mode on there.  I thought I did everything correct, but it looks like the widget is looking for an integer value and it's just text (ACRO, HOR, CRUZ, etc)  Am I missing some syntax in the sensors_maps.lua?

The error I get is ERROR in refresh(): /WIDGETS/YaapuMaps/lib/layout.lua:107: attempt to perform arithmetic on local 'sensorValue' (a string value

This is what I had:

[3]=  {
    "Mode",   -- label
    "FM",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)
  },

Other than my specific case, it's working FANTASTIC!  https://www.youtube.com/watch?v=QBl_wrMoskY

[image]
[image]
[image]

yaapu, 2025-01-09:
Hi, could you test this versione please?
[yaapumaps210_text_sensor.zip](https://github.com/user-attachments/files/18364719/yaapumaps210_text_sensor.zip)

rickgode, 2025-01-09:
Thanks!  I actually got a different error? ERROR in refresh(): WIDGETS/YaapuMaps/lib/maplib.lua;511;attempt to index field 'avgSpeed' (a nil value

Here is what I typed in to get the text sensor, maybe I mistyped something?
[image]

[3]=  {
    "Mode",   -- label
    "FM",     -- OpenTX sensor name
    0,          -- precision: number of decimals 0,1,2
    "",         -- label for unit of measure
    1,          -- multiplier if < 1 than divides
    "+",        -- "+" track max values, "-" track min values with
    2,          -- font size 1=small, 2=big
    nil,        -- warning level (nil is do not use feature)
    nil,        -- critical level (nil is do not use feature)

yaapu, 2025-01-09:
Hi, I cannot replicate it, could you save your telemetry so I can replay it in the simulator?

rickgode, 2025-01-09:
Sure!  I can't get the FM to show up in the simulator as it doesn't seem to be one of the sensors it discovers (plus I don't get that error in the sim)  
[flywooexplorer_sensors_maps.zip](https://github.com/user-attachments/files/18367580/flywooexplorer_sensors_maps.zip)
[Flywoo Explorer-2024-06-01-193929.csv](https://github.com/user-attachments/files/18367585/Flywoo.Explorer-2024-06-01-193929.csv)

Here is what it looks like in the sim for me as it doesn't discover the FM sensor
[image]

rickgode, 2025-04-08:
Anything new on this one?

yaapu, 2026-04-23:
@truglodite please try latest version, flight modes and generic text type sensors should work all right!

yaapu, 2026-04-24:
@truglodite cold restart should be no issue, did you manually reset home with a switch?
I can recheck the code but the widget waits for both lat/lon and home lat/lon to be not nil to start calculating, mhmmm

truglodite, 2026-04-24:
No I don’t have any reset home switches in my inav or my yaapu config. Such features have me scared that I might accidentally reset home miles away, and with my luck it will be the one flight where rth is actually needed, hehe.

FWIW, I’m running inav 9 with an m10 gps, and atomrc f405 Navi fc. I’m also running the latest edgetx and elrs releases.